vroland/epdiy #393: [Question] Decting broken/smashed epaper displays by TPS65185 VOM kick-back voltage?

cyberman54, 2025-01-23:
I'm looking for a way to detect a smashed / broken epaper display with epdiy, just by software.
The idea is: The TPS65185 can perform a voltage measurement on the VCOM pin to determine the kick-back voltage of the epd panel.
Would it be expectable to detect a smashed epaper display by reading this value and compare it against a certain threshold value?
Has anyone tested this?

martinberlin, 2025-01-24:
They won't be other way than to test this. From the TPS65185 datasheet we can read:

**8.3.7.1 Kick-Back Voltage Measurement**

TPS65185x can perform a voltage measurement on the VCOM pin to determine the kick-back voltage of the panel. This allows in-system calibration of VCOM. To perform a kick-back voltage measurement, follow these
steps:
• Pull the WAKEUP pin and the PWRUP pin high to enable all output rails.
• Set the HiZ bit in the VCOM2 register. This puts the VCOM pin in a high-impedance state.
• Drive the panel with the Null waveform. Refer to E-Ink specification for detail.
• Set the ACQ bit in the VCOM2 register to 1. This starts the measurement routine.
• When the measurement is complete, the ACQC (Acquisition Complete) bit in the INT1 register is set and the
nINT pin is pulled low.
• The measurement result is stored in the VCOM[8:0] bits of the VCOM1 and VCOM2 register.
The measurement result is not automatically programmed into nonvolatile memory. Changing the power-up default is described in the following paragraph.

Questions: 

1. What is the null waveform?
2. Would be the same just to refresh white the display and measure it?
3. Would be another means to detect vandalism like a sudden change of temperature (TPS65185 has also a thermistor) or simply that the Firmware will hang since cannot update the display. It all depends on how bad you call it broken. Because it can be completely smashed or just have a break in the side and in that case I'm not sure if the Kick-Back Voltage measurement will mark any difference. 
Hopefully will do if it's broken but also we don't know because it really needs some experimentation.

vroland, 2025-01-26:
Sounds like an interesting thing to try! We haven't even implemented kick-back measurement for normal VCOM calibration yet though, so that would be a prerequisite. And we know that there is quite a variance in VCOM from display to display, but detecting a change for the same display sounds pretty plausible. As martin said, you'll need to try ;)

cyberman54, 2025-01-26:
Yes, I'm going to try it out together with Martin with an experiment.

Is there a schematic diagram for epaper displays somewhere that shows the connection between Vcom, Vsource and Vgate?

martinberlin, 2025-01-26:
Not really there is no official schematic about this anywhere that I could find.
Here you can see one that was drawn by Petteri Aimonen 
[image]
So far we just know that VCOM is a surface reference voltage. Higher than 15 voltages are driving the pixel mosfet Gates supposedly one for letting pass -15 V (ink goes up) and another one gpr positive voltage per pixel.

cyberman54, 2025-01-26:
Thus, for breakage detection it may be useful to average Vcom reading over some time, while pulsing every single TFT FET.

cyberman54, 2025-01-27:
I asked eInk if this idea would work. They say yes, and that they applied this already in some applications. 
But they don't want to share details without NDA. 
So, let's DIY this here, and open source it :-)

martinberlin, 2025-01-27:
Awesome, is not a surprise Eink being so friendly and open, the first block to pass here in my opinion is:

- What is the NULL waveform. And how we can simulate that in epdiy?

To me it seems like a special waveform that is used for this VCOM kick-back voltage measurement. 

> Set the HiZ bit in the VCOM2 register. This puts the VCOM pin in a high-impedance state.

This point is already working in my demo. VCOM pin is in fact "in the air" and voltage is not assigned to it.

martinberlin, 2025-01-27:
Hello @zephray maybe you have a more clear idea of what is a "Null waveform" and how we could simulate it here?

cyberman54, 2025-01-27:
This [public Eink document](https://shopkits.eink.com/en/download/0223011714bb676200/User%20Manual%20-%2013.3''ePaper%20Display%20(VB3300-NCB).pdf) says there ist a "Mode 0 = INIT" Waveform:
"MODE 0 ─ INIT Initialize is used to completely clear the display, if it's left in an unknown
state (i.e. if the previous image has been lost by a re boot)"

martinberlin, 2025-01-27:
Thanks!   Yeah actually looks like a gray matrix (waveform) that is just clearing the display. Also WHITE, full black, and white again. I will try that in my next tests to see what comes out of the VCOM readings

zephray, 2025-01-27:
> Hello [@zephray](https://github.com/zephray) maybe you have a more clear idea of what is a "Null waveform" and how we could simulate it here?

The null waveform (based on my understanding) is just keep sending 0x00 (no operation) to the screen. This should allow doing VCOM kick back measurement

martinberlin, 2025-01-28:
Thanks a lot @zephray that helped a lot. Ok first tests are due and it's already reading something. Will play with this some days more anyways to see if it's the right way. @cyberman54 I've so far added 3 functions in tps65185 C files:

**void tps_vcom_kickback();**
 This one starts the board in this special mode:

```c
  // Pull the WAKEUP pin and the PWRUP pin high to enable all output rails.
  // But do not set VCOM to any voltage!
    epd_current_board()->measure_vcom(epd_ctrl_state());
  // PLUS Sets the HiZ bit in the VCOM2 register (BIT 5) 0x20
  // Adding BIT 3 & 4 Number of acquisitions that is averaged to a single kick-back V. measurement to 8 acq's
    tps_write_register(I2C_NUM_0, 4, 0x38);
```

**void tps_vcom_kickback_start();**

```c
    // Set the ACQ bit in the VCOM2 register to 1 (BIT 7)
    tps_write_register(I2C_NUM_0, TPS_REG_VCOM2, 0xA0); // BIT 5 + 7
    // IMPORTANT To keep in mind that we need to mind leaving other BITs in same state and not overwrite them, that's why here we also leave HiZ bit Set 
```

- unsigned tps_vcom_kickback_rdy();

Just checks if the ACQ bit is set. When it is then the readings should be readable from VCOM registers that we do with the following code:

```c
/**
 * @brief ACQC (Acquisition Complete) bit in the INT1 register is set
 * @return mV 0 is not read! 
 */
unsigned tps_vcom_kickback_rdy() {
    uint8_t int1reg = tps_read_register(I2C_NUM_0, TPS_REG_INT1);
    if (int1reg == 0x02) {
        uint8_t lsb = tps_read_register(I2C_NUM_0, 3);
        uint8_t msb = tps_read_register(I2C_NUM_0, 4);
        int u16Value = (lsb | (msb << 8)) & 0x1ff;
        printf("Kickback lsb:%d msb:%d result:%d\n\n", lsb, msb, u16Value);
        return u16Value *10;
    } else {
        return 0;
    }
}
```

In my tests so far I've used 2 displays and this is the value that comes out, taking 2 passes of the NULL waveform to get the readings:

1.  9.7" 1200x825, Vcom in the adhesive: -1780 mV.   VCOM Kickback reading in my test:    270 mV.  Disconnecting the display, readings are 0 like expected.
2. 11.5" display, Vcom in the adhesive: -1400 mV.  VCOM Kickback reading of 410 mV. 
Disconnecting the display, readings are 0 like expected.

Both readings where constant in my tests. Still didn't arrive to the part of breaking first displays since I want to make sure that the procedure is delivering a value we can use.
As a resume the VCOM readings I have are real. But they do not represent the VCOM that is on the display.  We are really not interested that they match because nowhere in the datasheet says it should be so. 
After more tries I will make first display broken but first I need to wait a week till a new one comes to my studio.

cyberman54, 2025-01-28:
Can you try the same approach, with some other waveform, while measuring? To see impact of waveform on voltage value. Also, could you try to heat or cool display why measuring, and compare value? To see impact of temperature on Vcom.

cyberman54, 2025-01-28:
Hmm, that's bad news because it makes things more complicated, meaning some kind of temperature compensation will be needed when comparing Vcom Values.

Solution: Always read Vcom and Temp value from TPS chip. Then combine both values in some way before compare.

Now depends on Vcom values we see with broken displays. If the variation is smaller than average temperature impact, a breakage detection via Vcom will be difficult or impossible, especially in outdoor environments.

martinberlin, 2025-01-28:
> Now depends on Vcom values we see with broken displays. If the variation is smaller than average temperature impact, a breakage detection via Vcom will be difficult or impossible, especially in outdoor environments.

Totally agree. In the cold winter nights under 0° C will be hard to detect. But also keep in mind I did this test putting the naked Display in the freezer. Such an extreme cold shock won't have when it's in the outdoor enclosure protected by glass.

But for for sure VCOM readings will be lower.  Tomorrow I will start breaking first a 6" that I've somewhere in the lab and don't use, and will document it in the WiKi, and in video. I'm hoping that when display is completely broken VCOM kickback will be close to 0.

cyberman54, 2025-01-28:
Not sure, if'm a right (i am NOT an electrical engineer): Regarding to the prinicple schematic on top, i would guess Vcom values will rise (not drop) when TFTs on the substrate break. If so, we have no trade off with temperature impact. Maybe we can even detect how many TFTs break and then guess "n% of display surface is broken". That would be extremly nice ;-) But theory so far.

martinberlin, 2025-01-28:
Between tomorrow and Friday I will make a real test of breaking and documenting a 6" display. 
For me the best way to measure VCOM in the open knowing that you can measure temperature, will be to do it always when temperature is >= 10°C also in the warmest moment in the day in winter (And possibly in the night on the summer)
Also try to make it only between 10 and 20°C so the drift as low as possible

martinberlin, 2025-01-28:
Additional note from my side:
The reason that VCOM is high if you measure it right afterwards doing a normal display operation with Display waveform is also that both VCOM and the pixel push +15 and -15 have 1 uF capacitors 

[image]

My theory is that then to do proper readings you need to wait some seconds until this capacitors drain. And then you apply the NULL waveform and read. This time VCOM is in HiZ state also like pin is in high impedance state, ready to get the voltage from the other layers. But it might be your theory is right and goes up when display substrate is broken, let's document it. I'm really curious to see what comes out

martinberlin, 2025-01-29:
@cyberman54 you where right!
Video proof: https://youtu.be/PyhPsOnWIcA (Is unlisted because my German is not good)

cyberman54, 2025-01-29:
> [@cyberman54](https://github.com/cyberman54) you where right! Video proof: https://youtu.be/PyhPsOnWIcA (Is unlisted because my German is not good)

Thank you very much for this experiment! Looks like a well paved road to go for a remote display health check! Lets go further now!

vroland, 2025-02-17:
Thanks for both of your work and @martinberlin for the implementation, I'll try it out on the weekend!

martinberlin, 2025-02-20:
About this @vroland I wanted to ask you for advice.

It seems if the display is badly broken it could also be the case that there is a short in the high voltage lines, which if you ignore it, and leave the PWRUP on high then it could badly damage the TPS65185 (You remember my bad experiment with the hair drier where I fried one IC)
According to datasheet: https://www.ti.com/lit/ds/symlink/tps65185.pdf **section 7.1**
Peak output current Internally limited   -- mA   Does not mention how much, but is internally limited, does not mean fully protected but making a short should not kill the IC

Operating junction temperature –10 **125 °C** --> And this is the most important if it heats up above that level then is fried. 

My proposed solution for external use-cases when display can be broken by vandalism or external factors would be:

**To make a Task that runs only when you activate power lines and that each second (or 2 seconds) checks that temperature does not pass 50 or 60°C.
If it does then there is an issue and it is over-heating. 
It will just cut power lines in TPS PMIC and report an error.**

Do you think it could be the right idea to implement?
Because actually won't change the behaviour, if any power line is grounded or tied to any other thing, effectively making the DC to DC extra limit himself and get hot, the display at this point is unusable anyways. So at least we can protect the TPS65185 IC to the point that only display has to be changed @cyberman54

closed 2025-03-16 by martinberlin