vroland/epdiy #404: Using the V7 board, the program stuck here, did not go down

ShiningMan123, 2025-03-11:
Using the V7 board, the program stuck here, did not go down

[image]

ShiningMan123, 2025-03-11:
E (64493) EPD: TPS write failed: -1
I (64603) epdiy: Using optimized vector implementation on the ESP32-S3, only 1k of 65536 LUT in use!
I (66493) epdiy: diff: 86ms, draw: 1866ms, buffer update: 24ms, total: 1976ms
I (68963) epdiy: Using optimized vector implementation on the ESP32-S3, only 1k of 65536 LUT in use!
I (70853) epdiy: diff: 86ms, draw: 1866ms, buffer update: 25ms, total: 1977ms
I (80863) EPD: Entering epd_board_poweron
I (80863) EPD: Before delay
I (80873) EPD: Before PWRGOOD check
I (80873) EPD: PWRGOOD check done, tries: 0
I (80873) EPD: Before TPS enable

martinberlin, 2025-03-11:
Not an issue with epdiy this is moved to Discussions now. Double check your hardware