vroland/epdiy #412: IDF build is giving errors on versions 5.3 to latest 5.5

martinberlin, 2025-03-29:
Hello @vroland 
Can you take a look if you find some minutes:
https://github.com/vroland/epdiy/blob/main/src/output_lcd/lcd_driver.c#L614

Function lcd_hal_cal_pclk_fre gives an error saying it’s one parameter too much.

To reproduce the error you need to pull latest updates from each version and try to build a v7 S3 board.
This is important since it will affect any newcomers when trying the epdiy v7 board with a fresh IDF installation >= 5.3

vroland, 2025-04-04:
Hm, I tried with the esp-idf-v5.3.2 release and it works for me. I noticed though that the docker container release/v5.3 has this problem, while release/v5.3.2 has not. How are you compiling this?

martinberlin, 2025-04-05:
To reproduce:
Just grab any official latest release. Let’s say you want a clean IDF 5.4 , completely remove all directory, clone repository and do a checkout of **release/v5.4**

For me with release/v5.3 release/v5.4 and release/v5.5 that is very close to master, all 3 are having this problem. So in my head if I’m thinking right, any newcomer that installs IDF to try this will hit this.
I don’t understand why it’s like this since this function use to have 5 parameters since v5.3 as it’s reflected in the codebase

martinberlin, 2025-04-11:
@vroland this is really easy to solve since the only extra parameter that should not be there is flags that is 0. The hard part is to know in what version to to add it and where not

lanmi42, 2025-04-14:
Hi,

I experienced this error with Arduino IDE as well.

A quick fix is to downgrade espressif esp32 board to 3.0.7

martinberlin, 2025-04-14:
There is already a fix proposed by @vroland in #413 
Seems to work great. Hope is merged soon!

lanmi42, 2025-04-14:
> There is already a fix proposed by [@vroland](https://github.com/vroland) in [#413](https://github.com/vroland/epdiy/pull/413) Seems to work great. Hope is merged soon!

Awesome! Thanks for letting me know.

closed 2025-04-28 by vroland