vroland/epdiy #417: LilyGo T5 4.7 Inch Example / miniz.h: No such file or directory

evanderkuy, 2025-05-05:
Hi,

I am trying to get my LilyGo T5 4.7 S3 (v2.3) to draw strings/texts. But I am stuck on the foloowing  error.:

 .pio/libdeps/esp32-s3-devkitc-1/epdiy/src/font.c:8:10: fatal error: miniz.h: No such file or directory

The repository that should provide the file no longer exists : https://github.com/nmoinvaz/miniz.git

PS : I can draw images using the example provided by Lilygo : https://github.com/Xinyuan-LilyGO/LilyGo-EPD47/tree/esp32s3/examples/drawImages

Please advise how to solve this.

Best regards
Etienne

evanderkuy, 2025-05-06:
Hi Martin,

Thank you for your fast reply. Do you mean when using Platformio I can still use the "LilyGo T5 4.7 Inch Example" (for Platformio) and compile/fash it with  idf toolchain?

I just tried to setup a Esp-idf project but that is too complicated for me, I am happy in Platformio (Arduino).

Etienne

martinberlin, 2025-05-07:
Yes I mean that probably PIO is using an older IDF and that's why you get this error.
Try to update Espressif package to the last version. And just try going to that directory in examples and do:

idf.py build
idf.py flash monitor

to see if that is the issue. With: idf.py --version   you can see what version is installed in your system

closed 2025-06-10 by martinberlin