vroland/epdiy #430: About esp32p4 compatibility issues

ShallowGreen123, 2025-07-01:
Hello, I am using esp32p4 chip to drive the display recently, but epdiy does not seem to be compatible with esp32p4 now. Will the author have plans to update to support esp32p4 in the future? Thank you very much.

vroland, 2025-07-20:
Hm, since this chip uses a different CPU architecture and likely quite different peripherals, it would likely be a lot of work to port. At the same time, It may not be a lot faster (The S3 version is very optimized with it's vector extensions) and the absence of wifi (maybe there are combo modules though, idk) makes it not very interesting for most use cases people use epdiy for.

martinberlin, 2025-11-30:
Moving this to Discussions - Ideas