vroland/epdiy #441: Failure to build with ESP-IDF v5.5 due to internal LCD APIs

nabilt, 2025-09-13:
I am having problems building the latest commit https://github.com/vroland/epdiy/commit/c61e9e923ce2418150d54f88cea5d196cdc40c54  with esp-idf v5.5

Build environment: Espressif docker docker v5.5 
docker run --rm -it --device=/dev/ttyUSB0 -v $PWD:/project -w /project espressif/idf:release-v5.5 bash

Build commands
```
cd examples/grayscale_test
idf.py set-target esp32s3
idf.py build
```

Output
```
In file included from /project/src/output_lcd/lcd_driver.c:42:
/project/src/output_lcd/lcd_driver.c: In function 'init_lcd_peripheral':
/opt/esp/idf/components/hal/esp32s3/include/hal/lcd_ll.h:723:15: error: '__DECLARE_RCC_ATOMIC_ENV' undeclared (first use in this function)
  723 |         (void)__DECLARE_RCC_ATOMIC_ENV; \
```

Looks like the internal LCD APIs being used require some kind of context lock. The commit referenced above indicates this was tested with esp-idf 5.5 so this is probably user error, but I'm not sure what I'm missing.

I also had an error before this where it couldn't find the component 'epdiy' so I removed the requirement from epdiy to get the code to build.

nabilt, 2025-09-14:
Thank you. I have that commit, but you caused me to look at my esp-idf history. Turns out release-v5.5 is an active branch. I should be on tag v5.5 which was created for the release. For anyone else the correct docker command is 

`docker run --rm -it --device=/dev/ttyUSB0 -v $PWD:/project -w /project espressif/idf:v5.5 bash`

I have some kind of watch dog issue now, but I'm not sure if this is because of my modifications. I'm using a dev board instead of the official v7 so I had to make changes for my setup. Logs are below in case this is a know issue, but I'll close this one and reopen if necessary. Thanks again. 

```
I (948) sleep_gpio: Configure to isolate all GPIO pins in sleep state
I (955) sleep_gpio: Enable automatic switching of GPIO sleep configuration
I (963) main_task: Started on CPU0
I (973) esp_psram: Reserving pool of 32K of internal memory for DMA/internal allocations
I (973) main_task: Calling app_main()
I (983) epdiy: using resolution 234x1408
ESP-ROM:esp32s3-20210327
Build:Mar 27 2021
rst:0x8 (TG1WDT_SYS_RST),boot:0x8 (SPI_FAST_FLASH_BOOT)
Saved PC:0x4038213e
--- 0x4038213e: xthal_save_extra_nw at ??:?
```

closed 2025-09-14 by nabilt

tyan-boot, 2025-10-29:
@jonasschnelli Hi, sorry for bother you, i think this issue is not fixed on #426, espressif introduce this `__DECLARE_RCC_ATOMIC_ENV` on this commit https://github.com/espressif/esp-idf/commit/9a2997e25cd73aa2126f51cd1b826b340cce433b#diff-417ae50e789693e018516d497733866846f0956ad4f3b09dce31ccc9d0eef3a1R712-R717 which only exist in v5.4.3 and above, so the CI checks should be ok on v5.4 but not on v5.4.3 and v5.5. 

The code from [M5GFX ](https://github.com/m5stack/M5GFX) might be a reference:
https://github.com/m5stack/M5GFX/blob/6c93c6836f11a5abad67c0ff35ff75050e636abc/src/lgfx/v1/platforms/esp32/common.cpp#L838