vroland/epdiy #442: ED115OC1 2760x2070 EPD Display Support?

kimasplund, 2025-09-17:
Does it support this 35€ display?

[image]

kimasplund, 2025-09-18:
Ah ok. But in general that connector works then with this board and existing connection? just the esp module thats to weak?

kimasplund, 2025-09-19:
Ok. Would it be possible to kobble on a esp32-p4 module?

[image]
[image]

martinberlin, 2025-09-20:
Nope. This project is not ready for P4. If you will do it great! Please keep us updated